kcjengr/qtpyvcp #1: ProbeBasic: ToolTable crashes when loading

KurtJacobson, 2019-01-10:
Reported by boban_serb on matrix: [linuxcnc.report.txt](https://github.com/kcjengr/qtpyvcp/files/2746568/linuxcnc.report.txt)

It seems ProbeBasic crashes when clicking on the Tool tab.

I can't reproduce this on my my machine, so it may have something to do with the data in the tool table file.

**Traceback:**
```
Traceback (most recent call last):
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/widgets/input_widgets/tool_table.py", line 190, in data
    return self._data(self.tool_list[index.row()], index.column(), role)
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/widgets/input_widgets/tool_table.py", line 213, in _data
    return item[3]
IndexError: list index out of range
```

bobanSerb, 2019-01-10:
I found why, if the tool.tbl file is empty, it causes the program crash

TurBoss, 2019-01-10:
I'll take a look

TurBoss, 2019-01-10:
I can't reproduce this issue

KurtJacobson, 2019-01-11:
@bobanSerb I can reproduce that now, thank you!
So it looks the error is caused if the D and offset values are missing.

KurtJacobson, 2019-01-11:
@bobanSerb I just pushed what I think is a fix. Can you test it?
Thanks!

closed 2019-01-11 by KurtJacobson

bobanSerb, 2019-01-12:
It's ok now , thanks.

commit 14cf9198 (2019-01-11): BUG: ToolTable - fix crash when loading (Fixes #1)

Fix tool table crash when loading from a tool table that does not have an offset values defined.