kcjengr/qtpyvcp #4: ImportError: No module named probe_basic_rc (Vertical Probe Basic Screen)

JackDaviesDesign, 2019-02-19:
(Please fill in this issue template with as much information as you can about the 
circumstances under which the issue occurred, and the steps needed to reproduce it.)

## Steps to reproduce the problem

(provide as detailed a step by step as you can) 

 1. Opened the Vertical Probe Basic screen

Normal Probe Basic Screen works fine.

## This is what I expected to happen

(explain what you thought should have happened)

## This is what happened instead

(explain what happened instead)

## It worked properly before this

(did it work before? what changed?)

## Traceback

```python
Traceback (most recent call last):
  File "/home/jacktest/.local/bin/qtpyvcp", line 11, in <module>
    load_entry_point('qtpyvcp==0.2.0+3.g358ef6e', 'console_scripts', 'qtpyvcp')()
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/__init__.py", line 88, in main
    run_vcp(opts, None)
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/__init__.py", line 111, in run_vcp
    load_vcp(opts)
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 124, in load_vcp
    _load_vcp_from_entry_point(vcp, opts)
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 169, in _load_vcp_from_entry_point
    vcp.main(opts)
  File "/home/jacktest/.local/lib/python2.7/site-packages/examples/probe_basic_vertical/__init__.py", line 19, in main
    qtpyvcp.run_vcp(opts, VCP_CONFIG_FILE)
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/__init__.py", line 138, in run_vcp
    launch_application(opts, config)
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 76, in launch_application
    loadWindows(config['windows'])
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 198, in loadWindows
    window = _initialize_object_from_dict(window_dict)
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 187, in _initialize_object_from_dict
    obj = _get_object_by_referance(provider)
  File "/home/jacktest/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 175, in _get_object_by_referance
    return getattr(importlib.import_module(modname), attrname)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/jacktest/.local/lib/python2.7/site-packages/examples/probe_basic_vertical/probe_basic.py", line 8, in <module>
    import probe_basic_rc
ImportError: No module named probe_basic_rc
```

## Options

```json
{
    "chooser": false, 
    "command_line_args": null, 
    "config_file": null, 
    "confirm_exit": null, 
    "fullscreen": false, 
    "help": false, 
    "hide_menu_bar": false, 
    "hide_status_bar": false, 
    "ini": "/home/jacktest/linuxcnc/configs/sim.qtpyvcp/xyz-metric.ini", 
    "log_file": "/home/jacktest/linuxcnc/configs/sim.qtpyvcp/sim.log", 
    "log_level": "INFO", 
    "maximize": false, 
    "perfmon": false, 
    "position": null, 
    "pref_file": null, 
    "qt_api": null, 
    "size": null, 
    "stylesheet": null, 
    "theme": null, 
    "vcp": "probe_basic_vertical", 
    "version": false
}
```

## System Info
```
 * Description:	Debian GNU/Linux 9.8 (stretch)
 * Kernel: 4.9.0-8-rt-amd64
 * Qt version: v5.7.1
 * Qt bindings: PyQt5 v5.7
 * LinuxCNC version: v2.8.0-pre1-4590-g584314c79
 * QtPyVCP version: v0.2.0+3.g358ef6e
```

## Attachments

Please also find and attach the following files, along with any others that may be helpful:
* /home/jacktest/linuxcnc/configs/sim.qtpyvcp/sim.log
* None

KurtJacobson, 2019-02-19:
Hey Jack, great to hear from you!

Try running `qcompile examples/probe_basic_vertical` from the qtpyvcp directory and see if that fixes the issue.

We have not been maintaining PB vertical and lathe very well lately. I'll see about updating them a bit.

JackDaviesDesign, 2019-02-19:
Hey Kurt!

Epic work with the screen, sorry I couldn't help out before, moved country to start a job (and brought my machine with me!) and took me a while to get back into it! Will check your suggestion tomorrow, thanks in advance. :)

leonpk82, 2019-02-21:
some of the status labels will need to be reset, a few have changed

closed 2021-01-25 by TurBoss