kcjengr/qtpyvcp #5: Acknowledgements to PyDM

hhslepicka, 2019-02-25:
Really nice Project here!

It would be great if you guys could add a nice Acknowledgements section at your project docs and/or README to PyDM (https://github.com/slaclab/pydm) given that a bunch of features such as Rules, the Extensions/TaskMenuExtension and a couple of other fundamental pieces were heavily based/copied from there!

The PyDM team would appreciate the recognition for all the effort put in coming up with the code, documentation and overall project structure.

@mattgibbs

hhslepicka, 2019-02-26:
Also, from the LICENSE file (https://github.com/slaclab/pydm/blob/master/LICENSE.md):

```
...
Redistribution and use in source and binary forms, with or without modification, are permitted provided that the following conditions are met:

(1) Redistributions of source code must retain the above copyright notice, this list of conditions and the following disclaimer.
...
```

This means that the LICENSE file should also be added at least at the files/code that was copied.

KurtJacobson, 2019-02-26:
@hhslepicka QtPyVCP has been immensely influenced by PyDM, not to mention that I learned most of what I know about Qt by reading the PyDM code base! Also I really appreciate the guidance and support you have provided on the PyDM gitter channel.

Currently the licence and file headers (and code in general) are a huge mess, but for now I will at least add/update the licence header for files that were directly influenced by PyDM, and add an acknowledgements section to the docs.

hhslepicka, 2019-02-26:
Thank you, @KurtJacobson . The PyDM Team appreciate it.
Feel free to reach out when needed.

commit c120afc5 (2019-03-07): Merge branch 'AckPyDM' (fixes #5)
closed 2019-03-07 by KurtJacobson