kcjengr/qtpyvcp #6: QPalette has no attribute inactive

Berkdan, 2019-03-28:
When Starting qtpyvcp with basic_probe interface the following error occurs then linuxcnc fails to start.

Traceback (most recent call last):
  File "/home/huseyin/.local/bin/qtpyvcp", line 11, in <module>
    load_entry_point('qtpyvcp==0.2.2+5.g0de9d84', 'console_scripts', 'qtpyvcp')()
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/__init__.py", line 89, in main
    run_vcp(opts, None)
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/__init__.py", line 112, in run_vcp
    load_vcp(opts)
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 126, in load_vcp
    _load_vcp_from_entry_point(vcp, opts)
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 171, in _load_vcp_from_entry_point
    vcp.main(opts)
  File "/home/huseyin/.local/lib/python2.7/site-packages/examples/probe_basic/__init__.py", line 19, in main
    qtpyvcp.run_vcp(opts, VCP_CONFIG_FILE)
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/__init__.py", line 139, in run_vcp
    launch_application(opts, config)
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 74, in launch_application
    loadWindows(config['windows'])
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 200, in loadWindows
    window = _initialize_object_from_dict(window_dict)
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/vcp_launcher.py", line 194, in _initialize_object_from_dict
    return obj(*args, **kwargs)
  File "/home/huseyin/.local/lib/python2.7/site-packages/examples/probe_basic/probe_basic.py", line 18, in __init__
    super(ProbeBasic, self).__init__(*args, **kwargs)
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/widgets/form_widgets/main_window.py", line 44, in __init__
    self.loadUi(ui_file)
  File "/home/huseyin/.local/lib/python2.7/site-packages/qtpyvcp/widgets/form_widgets/main_window.py", line 103, in loadUi
    uic.loadUi(ui_file, self)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/__init__.py", line 226, in loadUi
    return DynamicUILoader(package).loadUi(uifile, baseinstance, resource_suffix)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/Loader/loader.py", line 72, in loadUi
    return self.parse(filename, resource_suffix, basedir)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 1000, in parse
    actor(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 807, in createUserInterface
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 785, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 262, in createWidget
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 785, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 482, in createLayout
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 785, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 525, in handleItem
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 785, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 482, in createLayout
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 785, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 525, in handleItem
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 785, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 482, in createLayout
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 785, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 525, in handleItem
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 785, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 253, in createWidget
    self.stack.push(self.setupObject(widget_class, parent, elem))
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 219, in setupObject
    self.wprops.setProperties(obj, branch)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/properties.py", line 415, in setProperties
    prop_value = self.convert(prop, widget)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/properties.py", line 378, in convert
    return func(prop[0], **args)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/properties.py", line 284, in _palette
    sub_palette = getattr(QtGui.QPalette, palette_elem.tag.title())
AttributeError: type object 'QPalette' has no attribute 'inactive'

KurtJacobson, 2019-03-28:
@Berkdan thanks for the bug report!
Please post the output of this command: `python -c "import qtpy; print qtpy.QT_VERSION"`

Berkdan, 2019-03-28:
the output is
5.7.1

Thanks.

Berkdan, 2019-03-28:
If I need to give some more details,
I use Debian 9 (Stretch) with linuxcnc.
I installed QT designer first then
installed qtpyvcp with pip just like the document says then I run qtpyvcp/pyqt5designer/Qt5.7.1-64bit/install.sh 
I launched Linuxcnc configuration launcher then select sim.qtpyvcp --> xyz-metric
Then Probe Basic in VCP Chooser Window then click Launch VCP button
I get unhandled exception -AttributeError windows with the logs ı recently posted.

Thanks

Berkdan, 2019-04-09:
Isn't there any info about that? I am stuck. I tried lots of things to make it work but i get the same error everytime.

Berkdan, 2019-04-09:
ı got it working by deleting all inactive segments in probe_basic.ui file.

KurtJacobson, 2019-04-12:
@Berkdan Thank you for investigating the cause of this error! I was not able to duplicate the error here so glad you found out what was causing it. What do you mean by inactive segments in the UI file?

closed 2021-01-30 by TurBoss