kcjengr/qtpyvcp #9: DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.144 was not provided by any .service files

linden1910, 2019-04-14:
I am not sure what this is all about i get same error when i use the basic vcp tutorial as well.
any help or insight would be much appreciated 

thanks linden

Steps to reproduce the problem:

Down load plaz_beta from git hub https://github.com/linden1910/plaz_beta.git
pip install plaz_beta -e .  inside plaz_beta folder
copy plaz_beta folder from configs folder to linuxcnc configs folder
Start linuxcnc with plaz_beta.ini  linuxcnc '~/linuxcnc/configs/plaz_beta_sim/plaz_beta.ini'
Machine should start with manual tool changer message
Load gusset.ngc using file tab
select back plot tab
Jog machine and set part zero
press cycle start 
error appearers at first G1 move

## Traceback

```python
Traceback (most recent call last):
  File "/home/plaz/Git/Dev-QtpyVcp/qtpyvcp/qtpyvcp/plugins/notifications.py", line 134, in onTimeout
    self.error_message.setValue(msg)
  File "/home/plaz/Git/Dev-QtpyVcp/qtpyvcp/qtpyvcp/plugins/plugin.py", line 115, in setValue
    self.fset(self.instance, self, value)
  File "/home/plaz/Git/Dev-QtpyVcp/qtpyvcp/qtpyvcp/plugins/plugin.py", line 126, in inner
    fset(*args, **kwargs)
  File "/home/plaz/Git/Dev-QtpyVcp/qtpyvcp/qtpyvcp/plugins/notifications.py", line 83, in error_message
    self.captureMessage('error', msg)
  File "/home/plaz/Git/Dev-QtpyVcp/qtpyvcp/qtpyvcp/plugins/notifications.py", line 88, in captureMessage
    self.showDesktopNotification(typ, msg)
  File "/home/plaz/Git/Dev-QtpyVcp/qtpyvcp/qtpyvcp/plugins/notifications.py", line 104, in showDesktopNotification
    self.desktop_notifier.show(title=typ.capitalize(), body=msg, icon=icon)
  File "/home/plaz/Git/Dev-QtpyVcp/qtpyvcp/qtpyvcp/lib/notify.py", line 145, in show
    self.timeout,
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name :1.144 was not provided by any .service files
```

## Options

```json
{
    "chooser": false, 
    "command_line_args": null, 
    "config_file": null, 
    "confirm_exit": false, 
    "fullscreen": null, 
    "help": false, 
    "hide_menu_bar": false, 
    "hide_status_bar": false, 
    "ini": "/home/plaz/linuxcnc/configs/plaz_beta_sim/plaz_beta.ini", 
    "log_file": "/home/plaz/qtpyvcp.log", 
    "log_level": null, 
    "maximize": null, 
    "perfmon": false, 
    "position": null, 
    "pref_file": null, 
    "qt_api": null, 
    "size": null, 
    "stylesheet": null, 
    "theme": null, 
    "vcp": "plaz_beta", 
    "version": false
}
```

## System Info
```
 * Description:	Linux Mint 19.1 Tessa
 * Kernel: 4.9.0-8-rt-amd64
 * Qt version: v5.9.5
 * Qt bindings: PyQt5 v5.10.1
 * LinuxCNC version: v2.8.0~pre1
 * QtPyVCP version: v0.2.2+27.gc18f30e
```

## Attachments

Please also find and attach the following files

[qtpyvcp.log](https://github.com/kcjengr/qtpyvcp/files/3077171/qtpyvcp.log)
[gusset_2_inch_ngc.txt](https://github.com/kcjengr/qtpyvcp/files/3077173/gusset_2_inch_ngc.txt)

KurtJacobson, 2019-04-14:
@linden1910 Thanks for the very complete bug report!

Several people have mentioned this bug on the IRC channel, but I have not been able to reproduce it. My guess is that it is caused by the notifications server crashing for some reason. I think the best solution is to implement our own notification system. In the mean time you can remove the `data_plugins: notifications` entry from the `qtpyvcp/yaml_lib/default_config.yaml` file.

KurtJacobson, 2019-04-14:
BTW, just gave your VCP a spin, very nice layout! :+1:

linden1910, 2019-04-16:
Thanks Kurt

     It seems to work now. I will see if I can figure out whats going on 
and trouble shoot a bit more when I have some time.

We just brought our new baby boy back from the hospital yesterday so my 
spare time will be scarce in the coming weeks.

Thanks again for all the work you have put in to building this frame 
work to allow people like me with no programing back ground at all to 
build custom gui's to match our applications and machines.

Linden

KurtJacobson, 2019-04-16:
Absolutely! Congratulation on the baby boy! :baby: :tada:

closed 2021-01-25 by TurBoss