kcjengr/qtpyvcp #10: File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/vcp_chooser/vcp_chooser.py", line 82, in get_vcp_data     with open(vcp_config_file, 'r') as fh: IOError: [Errno 2] Datei oder Verzeichnis nicht gefunden: '/home/cnc/.local/lib/python2.7/site-packages/video_tests/qtpyvcp_test/config.yml' Shutting down and cleaning up LinuxCNC...

marfis89, 2019-04-24:
Hi guys, i tried to setup qtpyvcp. I think the setup was fine, but i got the following error message:
File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/vcp_chooser/vcp_chooser.py", line 82, in get_vcp_data
    with open(vcp_config_file, 'r') as fh:
IOError: [Errno 2] Datei oder Verzeichnis nicht gefunden: '/home/cnc/.local/lib/python2.7/site-packages/video_tests/qtpyvcp_test/config.yml'
Shutting down and cleaning up LinuxCNC...

marfis89, 2019-04-24:
if i manually add a config-yml file, i get the same error for vtk_test and then for the opengl folder

TurBoss, 2019-04-24:
hello,

could you please try to run this?
```
pip uninstall vtk_test
pip uninstall qtpyvcp_test
```
Thanks

marfis89, 2019-04-24:
Cannot uninstall requirement qtpyvcp-test, not installed
Cannot uninstall requirement vtk-test, not installed

if i try to install one of them, i get:
`Exception:`
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/lib/python2.7/dist-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'``

TurBoss, 2019-04-24:
if you try this it works?

```
linuxcnc sim/xyza.ini probebasic
```

KurtJacobson, 2019-04-24:
@marfis89 thanks for the bug report!
The video video test VCPs don't work when selected from the VCP chooser.

marfis89, 2019-04-24:
@KurtJacobson i cannot even start the VCP chooser. because the config.yml is missing

@TurBoss okay new error: qt widgets are missing...getting closer

TurBoss, 2019-04-24:
I had this issue when vcp chooser tried to load a vcp installed by pip during setup
but removed on qtpyvcp examples

i solved by uninstalling thoose vcps from pip with the uninstall command

TurBoss, 2019-04-24:
can you please attach the error messages you get?

thanks

marfis89, 2019-04-24:
@TurBoss i got
 ImportError: No module named 'PyQt5.QtQuick' 
i solved it by running:
 `sudo apt-get install python3-pyqt5.qtquick`

now i can manually start probe basic / brender with:
`linuxcnc sim.qtpyvcp/xyz.ini probebasic`

marfis89, 2019-04-25:
@TurBoss 
Im now able to control my cnc setup with probe basic <3 (great work btw ;-)) . Any idea how i can fix the yml.config error now? 

> i solved by uninstalling thoose vcps from pip with the uninstall command

I deleted any vcps i found so far! Other Places to serarch?

Which Monitor /Resulution would be perfect for probe Basic?

thanks

TurBoss, 2019-04-26:
> Im now able to control my cnc setup with probe basic <3 (great work btw ;-)) . Any idea how i can fix the yml.config error now?
>
> > i solved by uninstalling thoose vcps from pip with the uninstall command
>
> I deleted any vcps i found so far! Other Places to serarch?

could you please attach the output of run:
```
$linuxcnc sim/xyza.ini
```

> Which Monitor /Resulution would be perfect for probe Basic?

1920x1080

closed 2019-05-09 by jethornton