kcjengr/qtpyvcp #11: ImportError: No module named QtQuickWidgets

RENATOK, 2019-04-25:
(Please fill in this issue template with as much information as you can about the 
circumstances under which the issue occurred, and the steps needed to reproduce it.)

## Steps to reproduce the problem

(provide as detailed a step by step as you can) 

 1.
 2.
 3.

## This is what I expected to happen

(explain what you thought should have happened)

## This is what happened instead

(explain what happened instead)

## It worked properly before this

(did it work before? what changed?)

## Traceback

```python
Traceback (most recent call last):
  File "/home/renato/Downloads/kcjengr-qtpyvcp-bdff096/qtpyvcp/widgets/qtdesigner_plugin.py", line 38, in <module>
    from qtpyvcp.widgets.display_widgets.designer_plugins import *
  File "/home/renato/Downloads/kcjengr-qtpyvcp-bdff096/qtpyvcp/widgets/display_widgets/designer_plugins.py", line 46, in <module>
    from atc_widget.atc import DynATC
  File "/home/renato/Downloads/kcjengr-qtpyvcp-bdff096/qtpyvcp/widgets/display_widgets/atc_widget/atc.py", line 14, in <module>
    from qtpy.QtQuickWidgets import QQuickWidget
  File "/home/renato/.local/lib/python2.7/site-packages/qtpy/QtQuickWidgets.py", line 14, in <module>
    from PyQt5.QtQuickWidgets import *
ImportError: No module named QtQuickWidgets
```

## Options

```json
{}
```

## System Info
```
 * Description:	Debian GNU/Linux 9.8 (stretch)
 * Kernel: 4.9.0-8-rt-amd64
 * Qt version: v5.7.1
 * Qt bindings: PyQt5 v5.7
 * LinuxCNC version: v2.8.0-pre1-4701-g5d6f7e9a4
 * QtPyVCP version: 0+unknown
```

## Attachments

Please also find and attach the following files, along with any others that may be helpful:
* None
* None

TurBoss, 2019-04-26:
hello,

This is the package thats needed

```
$ sudo apt install python-pyqt5.qtquick
```

RENATOK, 2019-05-10:
Hi
Thank you for your help

closed 2019-05-10 by TurBoss