kcjengr/qtpyvcp #17: atc does not appear

RENATOK, 2019-05-12:
I have this problem I do not think the problem does anyone have any idea how to solve this?
below the dependencies that I have installed.
a picture of how the screen appears atc

[image]

[sudo] senha para renato: 
Lendo listas de pacotes... Pronto
Construindo árvore de dependências       
Lendo informação de estado... Pronto
git is already the newest version (1:2.11.0-3+deb9u4).
gstreamer1.0-plugins-bad is already the newest version (1.10.4-1).
pyqt5-dev-tools is already the newest version (5.7+dfsg-5).
python-dbus.mainloop.pyqt5 is already the newest version (5.7+dfsg-5).
python-pyqt5 is already the newest version (5.7+dfsg-5).
python-pyqt5.qtmultimedia is already the newest version (5.7+dfsg-5).
python-pyqt5.qtopengl is already the newest version (5.7+dfsg-5).
python-dev is already the newest version (2.7.13-2).
python-pip is already the newest version (9.0.1-2+deb9u1).
python-setuptools is already the newest version (33.1.1-1).
python-pyqt5.qsci is already the newest version (2.9.3+dfsg-4).
libqt5multimedia5-plugins is already the newest version (5.7.1~20161021-2).
Os seguintes pacotes foram instalados automaticamente e já não são necessários:
  icedtea-netx icedtea-netx-common libevent-2.0-5 linux-headers-4.9.0-8-amd64
  linux-headers-4.9.0-8-common
Utilize 'sudo apt autoremove' para os remover.
0 pacotes atualizados, 0 pacotes novos instalados, 0 a serem removidos e 1 não atualizados.

KurtJacobson, 2019-05-13:
Are there any errors in the terminal you run LCNC from?

TurBoss, 2019-05-13:
some dependencies are undocumented :( my fault

could you try to install 
```
qml-module-qtquick-controls2
```
thanks

KurtJacobson, 2019-05-13:
@RENATOK I think you are missing the QtQuick control widgets

`sudo apt install qml-module-qtquick-controls`

RENATOK, 2019-05-13:
Hi
Thank you all for your help!
I installed qml-module-qtquick-controls
but nothing happened!
in annex lcnc.
Thank you for your help  !

acceleration	1e+99
active_queue	0
actual_position	0.0000   0.0000   0.0000   0.0000   0.0000
adaptive_feed_enabled	False
ain	0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000
angular_units	1.0000
aout	0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000   0.0000
axes	5
axis_mask	31
block_delete	True
call_level	0
command	-
current_line	0
current_vel	0.0000
cycle_time	0.001
debug	0
delay_left	0.0
din	False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False
distance_to_go	0.0000
dout	False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False False
dtg	0.0000   0.0000   0.0000   0.0000   0.0000
echo_serial_number	13
enabled	True
estop	0
exec_state	done
feed_hold_enabled	True
feed_override_enabled	True
feedrate	1.0
file	-
flood	0
g5x_index	1
g5x_offset	0.2368   0.8603   -4.2555  0.0000   0.0000
g92_offset	0.0000   0.0000   0.0000   0.0000   0.0000
gcodes	G80 G17 G40 G20 G90 G94 G54 G49 G99 G64 G97 G91.1 G8
homed	1 1 1 1 1
id	0
inpos	True
input_timeout	False
interp_state	idle
interpreter_errcode	0
joints	5
kinematics_type	identity
limit	0 0 0 0 0
linear_units	0.0394
lube	1
lube_level	0
max_acceleration	999999999999999967336168804116691273849533185806555472917961779471295845921727862608739868455469056.0000
max_velocity	5.0000
mcodes	M5 M9 M48 M53 M0
mist	0
motion_line	0
motion_mode	teleop
motion_type	none
optional_stop	True
paused	False
pocket_prepped	-1
position	0.0000   0.0000   0.0000   -0.0000  -0.0000
probe_tripped	False
probe_val	0
probed_position	0.0000   0.0000   0.0000   0.0000   0.0000
probing	False
program_units	inch
queue	0
queue_full	False
queued_mdi_commands	0
rapidrate	1.0
read_line	0
rotation_xy	0.0000
settings	0.0000   10.0000  300.0000
spindle	0            direction 0
0         orient_state 0
0              enabled 0
0     override_enabled True
0                brake 1
0                homed False
0             override 1.0
0                speed 0.0
0         orient_fault 0
1            direction 0
1         orient_state 0
1              enabled 0
1     override_enabled False
1                brake 1
1                homed False
1             override 1.0
1                speed 0.0
1         orient_fault 0
2            direction 0
2         orient_state 0
2              enabled 0
2     override_enabled False
2                brake 1
2                homed False
2             override 1.0
2                speed 0.0
2         orient_fault 0
3            direction 0
3         orient_state 0
3              enabled 0
3     override_enabled False
3                brake 1
3                homed False
3             override 1.0
3                speed 0.0
3         orient_fault 0
4            direction 0
4         orient_state 0
4              enabled 0
4     override_enabled False
4                brake 1
4                homed False
4             override 1.0
4                speed 0.0
4         orient_fault 0
5            direction 0
5         orient_state 0
5              enabled 0
5     override_enabled False
5                brake 1
5                homed False
5             override 1.0
5                speed 0.0
5         orient_fault 0
6            direction 0
6         orient_state 0
6              enabled 0
6     override_enabled False
6                brake 1
6                homed False
6             override 1.0
6                speed 0.0
6         orient_fault 0
7            direction 0
7         orient_state 0
7              enabled 0
7     override_enabled False
7                brake 1
7                homed False
7             override 1.0
7                speed 0.0
7         orient_fault 0
spindles	1
state	rcs_done
task_mode	manual
task_paused	0
task_state	on
tool_in_spindle	0
tool_offset	0.0000   0.0000   0.0000   0.0000   0.0000
velocity	0.0

Lcvette, 2019-05-13:
what is your actual issue?

RENATOK, 2019-05-13:
Hi .
on the atc screen, I have only one white square shown in the attached picture

RENATOK, 2019-05-14:
Hi guys
I made little progress. the atc now appears the windget.
now I have these errors, it is probably missing copying the files or some installation failure
thanks in advance

[image]

TurBoss, 2019-05-15:
Hello,
you could  try to copy 

sim/subroutines

to 

linuxcnc/nc_files/subroutines

thanks

RENATOK, 2019-05-17:
Hi guys.
I was able to solve
thank you all !

closed 2019-05-17 by RENATOK