kcjengr/qtpyvcp #26: Problem with vtk pachake by installation

Gnevko, 2019-11-20:
Hi People, something is wrong with the installation process. I just tried to follow the instruction: 

https://qtpyvcp.kcjengr.com/install/installation.html  
 
`pip install --editable .
Obtaining file:///home/cnc/qtpyvcp
Collecting docopt (from qtpyvcp==0.2.7+64.g24c3338)
  Using cached https://files.pythonhosted.org/packages/a2/55/8f8cab2afd404cf578136ef2cc5dfb50baa1761b68c9da1fb1e4eed343c9/docopt-0.6.2.tar.gz
Collecting qtpy (from qtpyvcp==0.2.7+64.g24c3338)
  Using cached https://files.pythonhosted.org/packages/cd/fd/9972948f02e967b691cc0ca1f26124826a3b88cb38f412a8b7935b8c3c72/QtPy-1.9.0-py2.py3-none-any.whl
Collecting pyudev (from qtpyvcp==0.2.7+64.g24c3338)
  Using cached https://files.pythonhosted.org/packages/bc/a2/31a07829acea8e70a28c247f43fa5d981229ae0f9edfeddedf52de00709b/pyudev-0.21.0.tar.gz
Collecting psutil (from qtpyvcp==0.2.7+64.g24c3338)
  Using cached https://files.pythonhosted.org/packages/03/9a/95c4b3d0424426e5fd94b5302ff74cea44d5d4f53466e1228ac8e73e14b4/psutil-5.6.5.tar.gz
Collecting HiYaPyCo (from qtpyvcp==0.2.7+64.g24c3338)
  Using cached https://files.pythonhosted.org/packages/8c/1d/80ac5ac1d91eb3c6f7e7c14e3aacedefc5febb712a04977583e54747d414/HiYaPyCo-0.4.14-py2.py3-none-any.whl
Collecting pyopengl (from qtpyvcp==0.2.7+64.g24c3338)
  Using cached https://files.pythonhosted.org/packages/9c/1d/4544708aaa89f26c97cc09450bb333a23724a320923e74d73e028b3560f9/PyOpenGL-3.1.0.tar.gz
Collecting vtk (from qtpyvcp==0.2.7+64.g24c3338)
  Could not find a version that satisfies the requirement vtk (from qtpyvcp==0.2.7+64.g24c3338) (from versions: )
No matching distribution found for vtk (from qtpyvcp==0.2.7+64.g24c3338)`

TurBoss, 2019-11-21:
vtk can't be installed from pip on 686 machines
https://pypi.org/project/vtk/#files

Gnevko, 2019-11-22:
Thank you, you are right. But maybe it's better to check CPU architecture before the main installation process and display a corresponding message?

TurBoss, 2019-11-22:
its only mentioned in the docs
https://qtpyvcp.kcjengr.com/install/quick_start.html
but some warning like the vtk trhows is likely a good idea

this way it will not allow you to start install

TurBoss, 2019-11-22:
but you can workarround this by installing python-vtk or python-vtk6

TurBoss, 2020-02-08:
> but you can workaround this by installing python-vtk or python-vtk6 from apt

this is not true and the only way for now we know to install vtk is from pip and its only available for x64

closed 2020-02-08 by TurBoss