kcjengr/qtpyvcp #27: Display Info and Error Notifications

Gnevko, 2019-11-23:
Hi 

If I have a little bit more errors or infos let say as usual, than I can see only first(old) notifications in the list and the new notifications will not be displayed at all (see screen shot).

[image]

TurBoss, 2019-11-23:
I reversed the notifications order
thanks

Gnevko, 2019-11-23:
gut idea! but i think additionally will be better to display only new notifications, let say, received in last 5 minutes.

TurBoss, 2019-11-23:
not time but count looks better to me
so only show the latest 5 messages or something like that

Gnevko, 2019-11-23:
agree :)

TurBoss, 2019-12-07:
this was fixed on master will be out soon

Thanks for report

TurBoss, 2020-05-02:
this is fixed on 0.3.7
thanks

closed 2020-05-02 by TurBoss