kcjengr/qtpyvcp #28: CRITICAL: ImportError: No module named notify

Gnevko, 2019-12-07:
Today I just wonted to download and install the last version of qtpyvcp (development installation) with new tooltable slot, but after that the commands like:

editvcp gnevkovcp

or

editvcp probe_basic

fail with following error:

[code]
cnc@cnc:~$ editvcp gnevkovcp
Loading YAML config file: /home/cnc/gnevkovcp/gnevkovcp/config.yml
touch-probe.diameter-offset OrderedDict([('default_value', 0.0), ('persistent', True), ('min_value', 0), ('max_value', 1)])
Loading UI file: /home/cnc/gnevkovcp/gnevkovcp/ui/mainwindow.ui
Loading QSS file: /home/cnc/gnevkovcp/gnevkovcp/ui/style.qss

Starting QtDesigner ...

Resetting rcs_errors_printed because a new NML channel is being created.
[qtpyvcp-designer.vcp_launcher][CRITICAL]  Unhandled exception in qtdesigner_plugin.py line 36 (vcp_launcher.py:38)
Traceback (most recent call last):
  File "/home/cnc/qtpyvcp/qtpyvcp/widgets/qtdesigner_plugin.py", line 36, in <module>
    from qtpyvcp.widgets.form_widgets.designer_plugins import *
  File "/home/cnc/qtpyvcp/qtpyvcp/widgets/form_widgets/designer_plugins.py", line 28, in <module>
    from qtpyvcp.widgets.form_widgets.probe_widget.probe import SubCaller
  File "/home/cnc/qtpyvcp/qtpyvcp/widgets/form_widgets/probe_widget/probe.py", line 19, in <module>
    from qtpyvcp.lib.notify import Notification, Urgency, init
ImportError: No module named notify
updating units 0
[qtpyvcp-designer.vcp_launcher][CRITICAL]  Unhandled exception in qtdesigner_plugin.py line 36 (vcp_launcher.py:38)
Traceback (most recent call last):
  File "/home/cnc/qtpyvcp/qtpyvcp/widgets/qtdesigner_plugin.py", line 36, in <module>
    from qtpyvcp.widgets.form_widgets.designer_plugins import *
  File "/home/cnc/qtpyvcp/qtpyvcp/widgets/form_widgets/designer_plugins.py", line 28, in <module>
    from qtpyvcp.widgets.form_widgets.probe_widget.probe import SubCaller
  File "/home/cnc/qtpyvcp/qtpyvcp/widgets/form_widgets/probe_widget/probe.py", line 19, in <module>
    from qtpyvcp.lib.notify import Notification, Urgency, init
ImportError: No module named notify
[/code]

At the moment It's not possible to continue with GUI development - all qtpyvcp widgets are not available in the editor.

TurBoss, 2019-12-07:
Fixed on master

Gnevko, 2019-12-07:
It works now, thank you!

closed 2019-12-07 by Gnevko