kcjengr/qtpyvcp #42: Wrong preview of rounded rectangle path in probe-basic.

rmu75, 2020-06-30:
I'm not sure if this is the proper place to report that, I think this is not really a problem of probe basic, but a problem in the vtk backplot.

Description of problem is here https://github.com/LinuxCNC/linuxcnc/issues/892

TurBoss, 2020-06-30:
Hello, 

I took the file Jepler uploaded and it runs fine on a metric sim

[image]

TurBoss, 2020-06-30:
G92 is not a well tested feature

rmu75, 2020-06-30:
> I took the file Jepler uploaded and it runs fine on a metric sim

I have no doubt that the file displays and works fine under normal circumstances. Can you check what happens with G92 offset(s)?

TurBoss, 2020-06-30:
sure will take a look tonight

TurBoss, 2020-06-30:
Hello, I just created a branch called "VTK_G92" in https://github.com/kcjengr/qtpyvcp
it should fix load files with a G92 set

TurBoss, 2020-07-05:
hello, 
did you have the opportunity test the fix?

Thanks

rmu75, 2020-09-08:
Sorry for the late reply. Just tested it, and it seems to work.

TurBoss, 2020-09-08:
no problem 
thanks for testing

i have to investigate this in deep because I don't like the way i did
it messes with having multiple parts in g5x and didn't expend much more time on that 

i have other branch wich went worse...

so the thing is on the air

TurBoss, 2021-03-18:
thanks to @85vmh this issue is fixed now

I'm closing this issue

thank you

closed 2021-03-18 by TurBoss