kcjengr/qtpyvcp #50: ImportError: No module named probe_basic_rc

JonoGee, 2020-09-26:
(Please fill in this issue template with as much information as you can about the 
circumstances under which the issue occurred, and the steps needed to reproduce it.)

## Steps to reproduce the problem

(provide as detailed a step by step as you can) 

 1.Fresh 2.8 LinuxCNC install
 2. Install Probe_basic as per instructions
 3.Ran Probe Basic

Got this error message.
Never had it working before. 1st installation attempt.

## Traceback

```python
Traceback (most recent call last):
  File "/home/cnc/.local/bin/probe_basic", line 11, in <module>
    load_entry_point('probe-basic', 'gui_scripts', 'probe_basic')()
  File "/home/cnc/probe_basic/probe_basic/__init__.py", line 20, in main
    qtpyvcp.run_vcp(opts, VCP_CONFIG_FILE)
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/__init__.py", line 27, in run_vcp
    run(*args, **kwargs)
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/app/__init__.py", line 123, in run
    launch_application(opts, config)
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/app/launcher.py", line 78, in launch_application
    loadWindows(config['windows'])
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/app/launcher.py", line 228, in loadWindows
    window = _initialize_object_from_dict(window_dict)
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/app/launcher.py", line 203, in _initialize_object_from_dict
    obj = _get_object_by_referance(provider)
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/app/launcher.py", line 191, in _get_object_by_referance
    return getattr(importlib.import_module(modname), attrname)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/cnc/probe_basic/probe_basic/probe_basic.py", line 12, in <module>
    import probe_basic_rc
ImportError: No module named probe_basic_rc
```

## Options

```json
{
    "command_line_args": null, 
    "config_file": "/home/cnc/linuxcnc/configs/probe_basic/probe_basic/custom_config.yml", 
    "confirm_exit": false, 
    "develop": false, 
    "fullscreen": null, 
    "help": false, 
    "hide_cursor": false, 
    "hide_menu_bar": false, 
    "hide_status_bar": false, 
    "info": false, 
    "ini": "/home/cnc/linuxcnc/configs/probe_basic/probe_basic/probe_basic.ini", 
    "log_file": "/home/cnc/linuxcnc/configs/probe_basic/probe_basic/sim.log", 
    "log_level": null, 
    "maximize": null, 
    "perfmon": false, 
    "position": "0x0", 
    "qt_api": null, 
    "size": null, 
    "stylesheet": null, 
    "theme": null, 
    "version": false
}
```

## System Info
```
 * Description:	Debian GNU/Linux 10 (buster)
 * Kernel: 4.19.0-10-rt-amd64
 * Qt version: v5.11.3
 * Qt bindings: PyQt5 v5.11.3
 * LinuxCNC version: v2.8.0
 * QtPyVCP version: v0.3.9
```

## Attachments

Please also find and attach the following files, along with any others that may be helpful:
* /home/cnc/linuxcnc/configs/probe_basic/probe_basic/sim.log
* /home/cnc/linuxcnc/configs/probe_basic/probe_basic/custom_config.yml

TurBoss, 2020-09-26:
Hello,
you need to compile the resource file maybe skiped by old installer
can you issue this commands
```
cd /home/cnc/.local/lib/python2.7/site-packages/probe_basic/
qcompile .
```

Thanks

JonoGee, 2020-09-27:
Thank you. I managed to get this working ok. Do you know if its possible to change this UI to POrtrait instead of landscape?

TurBoss, 2020-09-27:
the Portrait or vertical version still work in progress

TurBoss, 2020-09-30:
hello,

since the issue was fixed in master I'm closing this thread

thanks for the feedback

closed 2020-09-30 by TurBoss