kcjengr/qtpyvcp #51: Errors behind main GUI with Mate window manager

sttts, 2020-09-27:
The error popups show up behind the qtpyvcp main (fullscreen) window, i.e. invisible in normal operations.

Debian 10 Buster with Mate and marco window manager.

TurBoss, 2020-09-30:
Hello,

the fullscreen mode has some issues with the notification popups they can't go more levels up than the full screen

TurBoss, 2020-10-25:
Hello, 

sorry for the late reply
this may be fixed in #56 

Thanks for the report

sttts, 2020-10-27:
Doesn't look it is fixed.

Talking about these notifications:

[image]

sttts, 2020-10-28:
These are linuxcnc widgets, aren't they? I.e. not Qt and hence with a different parent window.

TurBoss, 2020-10-28:
Thanks!
I managed to fix the notifications :partying_face: 

thoose fixes are on latest master

>These are linuxcnc widgets, aren't they? I.e. not Qt and hence with a different parent window.

The native notifications which come configured on ProbeBasic are made with QWtidgets
There is a option in custom_comfig.yml that allows you to switch to DBus based notifications but doesn't work on all systems

https://github.com/kcjengr/probe_basic/blob/master/config/probe_basic/custom_config.yml
something like...
```yaml

data_plugins:
  notifications:
    provider: qtpyvcp.plugins.notifications:Notifications
    kwargs:
      enabled: True
      mode: native | dbus

```
and again thanks for the feddback!

sttts, 2020-10-28:
With dbus I now get visible notifications. Unfortunately, only one. If there are two or more notifications only the last one is visible.

TurBoss, 2020-11-02:
trigger notifications by hand does the same... only the latest is shown

```
$ notify-send "Title" "Message"

sttts, 2020-11-02:
Compare https://github.com/kcjengr/qtpyvcp/tree/Notifictions. There @TurBoss prototyped setting the notification widget's parent to the main window. This works for me with Mate window manager.

TurBoss, 2020-11-02:
I only need to ask @KurtJacobson if this is the correct place to init post main window

https://github.com/kcjengr/qtpyvcp/compare/Notifictions#diff-94fc3bc6e7fc8a770ba01b4c3b7d5117a29f10243fb0eb988bf0b4ebee3bcdd2R254

TurBoss, 2020-11-06:
done thanks for the feedback

PR #56 (2020-10-27): All dialogs stay on top
fixes #51
closed 2020-11-06 by TurBoss