kcjengr/qtpyvcp #64: IOError: [Errno 13] Keine Berechtigung: u'/home/dieter/linuxcnc/nc_files/examples/Untitled.ngc'

Herican, 2020-12-29:
(Please fill in this issue template with as much information as you can about the 
circumstances under which the issue occurred, and the steps needed to reproduce it.)

## Steps to reproduce the problem

(provide as detailed a step by step as you can) 

 1.
 2.
 3.

## This is what I expected to happen

(explain what you thought should have happened)

## This is what happened instead

(explain what happened instead)

## It worked properly before this

(did it work before? what changed?)

## Traceback

```python
Traceback (most recent call last):
  File "/home/dieter/.local/lib/python2.7/site-packages/qtpyvcp_conversational_gcode/widgets/base_widget.py", line 155, in on_post_to_file
    f.write_to_file(program_path)
  File "/home/dieter/.local/lib/python2.7/site-packages/qtpyvcp_conversational_gcode/ops/gcode_file.py", line 41, in write_to_file
    f = open(filename, 'w')
IOError: [Errno 13] Keine Berechtigung: u'/home/dieter/linuxcnc/nc_files/examples/Untitled.ngc'
```

## Options

```json
{
    "command_line_args": null, 
    "config_file": null, 
    "confirm_exit": false, 
    "develop": false, 
    "fullscreen": true, 
    "help": false, 
    "hide_cursor": false, 
    "hide_menu_bar": false, 
    "hide_status_bar": false, 
    "info": false, 
    "ini": "/home/dieter/linuxcnc/configs/Probe_Basic_Dieter/Probe_Basic_Dieter.ini", 
    "log_file": "/home/dieter/linuxcnc/configs/Probe_Basic_Dieter/sim.log", 
    "log_level": "WARN", 
    "maximize": true, 
    "perfmon": false, 
    "position": "0x0", 
    "qt_api": null, 
    "size": null, 
    "stylesheet": null, 
    "theme": null, 
    "version": false
}
```

## System Info
```
 * Description:	Debian GNU/Linux 10 (buster)
 * Kernel: 4.19.0-13-rt-amd64
 * Qt version: v5.11.3
 * Qt bindings: PyQt5 v5.11.3
 * LinuxCNC version: v2.8.1-13-ga22256354
 * QtPyVCP version: v0.3.11+12.g0923dcdf
```

## Attachments

Please also find and attach the following files, along with any others that may be helpful:
* /home/dieter/linuxcnc/configs/Probe_Basic_Dieter/sim.log
* None

TurBoss, 2021-01-25:
Hello,
looks like a permission issue 
the examples folder is a sim-link to the systems linuxcnc installation

it should work fine in any other directory where you have read and write permissions

an dialog or popup with the "permission denied" message can be adder

Thanks for the report

joco-nz, 2023-05-05:
No further activity in~2 yrs so closing.

closed 2023-05-05 by joco-nz