kcjengr/qtpyvcp #68: AttributeError: 'NoneType' object has no attribute 'setNotify'

gesamtplan, 2021-01-18:
(Please fill in this issue template with as much information as you can about the 
circumstances under which the issue occurred, and the steps needed to reproduce it.)

## Steps to reproduce the problem

(provide as detailed a step by step as you can) 

 1.
 2.
 3.

## This is what I expected to happen

(explain what you thought should have happened)

## This is what happened instead

(explain what happened instead)

## It worked properly before this

(did it work before? what changed?)

## Traceback

```python
Traceback (most recent call last):
  File "/home/charon/.local/lib/python2.7/site-packages/qtpyvcp/plugins/notifications.py", line 126, in timerEvent
    self.captureMessage('error', msg)
  File "/home/charon/.local/lib/python2.7/site-packages/qtpyvcp/plugins/notifications.py", line 82, in captureMessage
    self.notification_dispatcher.setNotify(m_type, msg)
AttributeError: 'NoneType' object has no attribute 'setNotify'
```

## Options

```json
{
    "chooser": false, 
    "command_line_args": null, 
    "config_file": "/home/charon/linuxcnc/configs/sim.qtpyvcp/custom_config.yml", 
    "confirm_exit": false, 
    "develop": false, 
    "fullscreen": null, 
    "help": false, 
    "hide_cursor": false, 
    "hide_menu_bar": false, 
    "hide_status_bar": false, 
    "info": false, 
    "ini": "/home/charon/linuxcnc/configs/sim.qtpyvcp/xyz-metric.ini", 
    "log_file": "/home/charon/linuxcnc/configs/sim.qtpyvcp/sim.log", 
    "log_level": "WARN", 
    "maximize": null, 
    "perfmon": false, 
    "position": null, 
    "qt_api": null, 
    "size": null, 
    "stylesheet": null, 
    "theme": null, 
    "vcp": "brender", 
    "version": false
}
```

## System Info
```
 * Description:	Debian GNU/Linux 10 (buster)
 * Kernel: 4.19.0-11-rt-amd64
 * Qt version: v5.11.3
 * Qt bindings: PyQt5 v5.11.3
 * LinuxCNC version: v2.8.1
 * QtPyVCP version: v0.3.11
```

## Attachments

Please also find and attach the following files, along with any others that may be helpful:
* /home/charon/linuxcnc/configs/sim.qtpyvcp/sim.log
* /home/charon/linuxcnc/configs/sim.qtpyvcp/custom_config.yml

gesamtplan, 2021-01-18:
Got the same Issue as sebastianm1986. Are the changes made four days ago checked in in pip. Installed the package over pypy. The version history shows the last package is from Nov 15th.

joco-nz, 2023-05-05:
Closing as old and part of 2.7 branch which is now defunct as Python 2 is end of life.

closed 2023-05-05 by joco-nz