kcjengr/qtpyvcp #72: DRO widgets line edit resets on tiny changes of actual position on machines with feedback

FranseFrikandel, 2021-03-03:
The new DRO widgets that allow mathmatical expressions to be filled in for DRO widgets are hard to use on machines that use feedback encoders, since these generally move about a tiny bit. A movement of 0.001mm, which in my case is only about a single encoder pulse, already causes the labels to reset whatever you typed in them.

In the attached video the issue is shown with probe_basic. I have already increased the update time in the YML file from 15ms to 150ms, yet this still makes it very hard to properly use the DRO widget as intended.

https://user-images.githubusercontent.com/30059330/109824417-1ad3cb80-7c39-11eb-9f53-80a5705ca563.mov

KurtJacobson, 2021-03-03:
@FranseFrikandel yes, that is definitely an issue!
A good, simple to implement solution would be to pause updates if the line edit has focus.

Will work on this.

FranseFrikandel, 2021-03-03:
Possible resolution might be to stop updating the DRO widget when a user enters something, until either the user confirms and the mathmatical expression is executed, or the commanded position gets changed.

FranseFrikandel, 2021-03-03:
> 
> 
> @FranseFrikandel yes, that is definitely an issue!
> A good, simple to implement solution would be to pause updates if the line edit has focus.
> 
> Will work on this.

Only potential issue I see in this is when a user with MPG clicks the line edit, then jogs his edgedfinder. The value in the widget will not be updated. Hence why I think forcing an update on change of the commanded axis position should probably be added.

FranseFrikandel, 2021-03-05:
Just wrote a potential fix. Will test on my own machine tomorrow and if it works will make a pull request for it.

closed 2021-03-26 by FranseFrikandel