kcjengr/qtpyvcp #75: AttributeError: 'NoneType' object has no attribute 'setNotify'

dadoblu2000, 2021-03-17:
(Please fill in this issue template with as much information as you can about the 
circumstances under which the issue occurred, and the steps needed to reproduce it.)

## Steps to reproduce the problem

(provide as detailed a step by step as you can) 

 1.
 2.
 3.

## This is what I expected to happen

(explain what you thought should have happened)

## This is what happened instead

(explain what happened instead)

## It worked properly before this

(did it work before? what changed?)

## Traceback

```python
Traceback (most recent call last):
  File "/home/linuxcnc/.local/lib/python2.7/site-packages/qtpyvcp/plugins/notifications.py", line 126, in timerEvent
    self.captureMessage('error', msg)
  File "/home/linuxcnc/.local/lib/python2.7/site-packages/qtpyvcp/plugins/notifications.py", line 82, in captureMessage
    self.notification_dispatcher.setNotify(m_type, msg)
AttributeError: 'NoneType' object has no attribute 'setNotify'
```

## Options

```json
{
    "chooser": false, 
    "command_line_args": null, 
    "config_file": null, 
    "confirm_exit": null, 
    "develop": false, 
    "fullscreen": null, 
    "help": false, 
    "hide_cursor": false, 
    "hide_menu_bar": false, 
    "hide_status_bar": false, 
    "info": false, 
    "ini": "/home/linuxcnc/linuxcnc/configs/probe_basic/config/atc_sim/vmc_index.ini", 
    "log_file": "/home/linuxcnc/qtpyvcp.log", 
    "log_level": null, 
    "maximize": null, 
    "perfmon": false, 
    "position": null, 
    "qt_api": null, 
    "size": null, 
    "stylesheet": null, 
    "theme": null, 
    "vcp": "probe_basic", 
    "version": false
}
```

## System Info
```
 * Description:	Debian GNU/Linux 10 (buster)
 * Kernel: 4.19.0-14-rt-amd64
 * Qt version: v5.11.3
 * Qt bindings: PyQt5 v5.11.3
 * LinuxCNC version: v2.8.1
 * QtPyVCP version: v0.3.12
```

## Attachments

Please also find and attach the following files, along with any others that may be helpful:
* /home/linuxcnc/qtpyvcp.log
* None

FranseFrikandel, 2021-03-18:
Duplicate of #68.

Presumably this happens upon startup? Perhaps try testing with a different VCP just to eliminate it actually being a probe_basic bug.

What appears to happen is that postGuiInitialise() of the Notifications dataplugin is not being called. I don't know how or why this would happen though.

dadoblu2000, 2021-03-24:
Error report created by /usr/lib/tcltk/linuxcnc/show_errors.tcl:

Print file information:
RUN_IN_PLACE=no
LINUXCNC_DIR=
LINUXCNC_BIN_DIR=/usr/bin
LINUXCNC_TCL_DIR=/usr/lib/tcltk/linuxcnc
LINUXCNC_SCRIPT_DIR=
LINUXCNC_RTLIB_DIR=/usr/lib/linuxcnc/modules
LINUXCNC_CONFIG_DIR=
LINUXCNC_LANG_DIR=/usr/lib/tcltk/linuxcnc/msgs
INIVAR=inivar
HALCMD=halcmd
LINUXCNC_EMCSH=/usr/bin/wish8.6
LINUXCNC - 2.8.1
Machine configuration directory is '/home/cnc/linuxcnc/configs/probe_basic/config/probe_basic'
Machine configuration file is 'probe_basic.ini'
INIFILE=/home/cnc/linuxcnc/configs/probe_basic/config/probe_basic/probe_basic.ini
VERSION=1.1
PARAMETER_FILE=sim.var
TASK=milltask
HALUI=halui
DISPLAY=probe_basic
COORDINATES=X Y Z A B
KINEMATICS=trivkins coordinates=XYZAB
Starting LinuxCNC...
Starting LinuxCNC server program: linuxcncsvr
Loading Real Time OS, RTAPI, and HAL_LIB modules
Starting LinuxCNC IO program: io
Starting HAL User Interface program: halui
twopass:invoked with <> options
twopass:found ./hallib/core_sim_5.hal
twopass:found ./hallib/spindle_sim.hal
twopass:found ./hallib/simulated_home_probe_basic.hal
twopass:found ./hallib/sim_cannon.hal
Starting TASK program: milltask
Starting DISPLAY program: probe_basic
Shutting down and cleaning up LinuxCNC...
task: 589 cycles, min=0.000082, max=0.056105, avg=0.009638, 0 latency excursions (> 10x expected cycle time of 0.010000s)
File: tool.tbl Unrecognized line skipped: ;Tool  Pocket Z Offset     Diameter     Remark

Removing HAL_LIB, RTAPI, and Real Time OS modules
Removing NML shared memory segments

Debug file information:
Note: Using POSIX realtime
motion.c: Creating unlock hal pins for joint 4
ERROR: In /work/standalone-x64-build/VTK-source/Rendering/OpenGL2/vtkOpenGLRenderWindow.cxx, line 797
vtkXOpenGLRenderWindow (0x559cb9784830): GL version 2.1 with the gpu_shader4 extension is not supported by your graphics driver but is required for the new OpenGL rendering backend. Please update your OpenGL driver. If you are using Mesa please make sure you have version 10.6.5 or later and make sure your driver in Mesa supports OpenGL 3.2.

ERROR: In /work/standalone-x64-build/VTK-source/Rendering/OpenGL2/vtkOpenGLRenderWindow.cxx, line 797
vtkXOpenGLRenderWindow (0x559cbc281410): GL version 2.1 with the gpu_shader4 extension is not supported by your graphics driver but is required for the new OpenGL rendering backend. Please update your OpenGL driver. If you are using Mesa please make sure you have version 10.6.5 or later and make sure your driver in Mesa supports OpenGL 3.2.

/usr/bin/linuxcnc: riga 896:  2319 Errore di segmentazione $EMCDISPLAY -ini "$INIFILE" $EMCDISPLAYARGS $EXTRA_ARGS
2283
2315
Stopping realtime threads
Unloading hal components
Note: Using POSIX realtime

-----------------------------------------------------------------------
Info report created by linuxcnc_info:
The file:    /tmp/linuxcnc_info.txt
can be posted to a forum or a web site like:
     http://pastebin.com
in order to provide information about the linuxcnc
system and configuration.

                Date: mer 24 mar 2021, 12.17.50, CET
            UTC Date: mer 24 mar 2021, 11.17.50, UTC
[... 42 lines cut ...]
               glade: not_in_PATH
          glade-gtk2: not_in_PATH

linuxcnc_var all:

     LINUXCNCVERSION: 2.8.1
LINUXCNC_AUX_GLADEVCP: /usr/share/linuxcnc/aux_gladevcp
LINUXCNC_AUX_EXAMPLES: /usr/share/linuxcnc/aux_examples
            REALTIME: /etc/init.d/realtime
                 RTS: uspace
          HALLIB_DIR: /usr/share/linuxcnc/hallib

hi, this is the error report when i try to start probe-basic 

dpkg -l '*linuxcnc*':
Voluto=U (non noto)/I (installato)/R (rimosso)/P (rimosso totale)/H (in attesa)
| Stato=Non/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(nessuno)/R (reinstallazione richiesta) (Stato,Err: maiuscolo=grave)
||/ Nome            Versione     Architettura Descrizione
+++-===============-============-============-=====================================================================
un  linuxcnc        <nessuna>    <nessuna>    (nessuna descrizione disponibile)
un  linuxcnc-doc    <nessuna>    <nessuna>    (nessuna descrizione disponibile)
ii  linuxcnc-doc-en 1:2.8.1      all          motion controller for CNC machines and robots (English documentation)
un  linuxcnc-sim    <nessuna>    <nessuna>    (nessuna descrizione disponibile)
ii  linuxcnc-uspace 1:2.8.1      amd64        motion controller for CNC machines and robots

FranseFrikandel, 2021-03-25:
As mentioned in LinuxCNC discord, relevant error message here is:

`ERROR: In /work/standalone-x64-build/VTK-source/Rendering/OpenGL2/vtkOpenGLRenderWindow.cxx, line 797
vtkXOpenGLRenderWindow (0x559cb9784830): GL version 2.1 with the gpu_shader4 extension is not supported by your graphics driver but is required for the new OpenGL rendering backend. Please update your OpenGL driver. If you are using Mesa please make sure you have version 10.6.5 or later and make sure your driver in Mesa supports OpenGL 3.2.`

You will need to update graphics drivers. How this is done depends on hardware and what exact distro you're running.

dadoblu2000, 2021-03-25:
I've been trying to figure out how to update video drivers for days but I can't get a spider out of the hole, can you help me? debian operating system 10 

glxinfo | grep OpenGL

OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Q35 
OpenGL version string: 1.4 Mesa 18.3.6
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 18.3.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
OpenGL ES profile extensions:

is this information correct?

FranseFrikandel, 2021-03-25:
Seems odd. Your CPU does not appear to have integrated graphics, so I assume OpenGL vendor should not be Intel, but rather AMD/ATI or NVidia if you were running a dedicated GPU.

dadoblu2000, 2021-03-26:
the graphics are integrated I have no video cards added in the PCIe or PCI slots

dadoblu2000, 2021-03-26:
I add log files for any checks, I just want to know if the video drivers are upgradeable to OpenGL 3.3 or if I need to add an alternative video card

[log_26mar_095428.txt](https://github.com/kcjengr/qtpyvcp/files/6210559/log_26mar_095428.txt)

FranseFrikandel, 2021-03-26:
Dont know much about this type of stuff. But managed to find it was loading i915 drivers for integrated graphics, and then found this: https://wiki.archlinux.org/index.php/intel_graphics#OpenGL_2.1_with_i915_driver which might resolve your issue.

Websites I find seem to suggest you would have a Radeon HD2400 Pro built in but there is no mention of it in the logs so I assume this is wrong.

dadoblu2000, 2021-03-26:
I immediately go to read, to me it reports this description related to the video card
VGA compatible controller: Intel Corporation 82Q35 Express Integrated Graphics Controller (rev 02)

dadoblu2000, 2021-03-26:
I was able to read and then verify the existence of the folder mentioned in the article, but / drirc does not exist in my operating system

FranseFrikandel, 2021-04-03:
I had sent you a message in Discord, but just noticed you haven't seen it yet, so I might have the wrong account or something.

I believe you can just create the drirc folder if it doesn't exist.

dadoblu2000, 2021-04-12:
Hello sorry if I did not answer, but I have been busy with work, I think the problem is hardware and I need to add a video card that supports opengl, so they told me from a forum but I don't know what to do

closed 2023-05-05 by joco-nz