kcjengr/qtpyvcp #96: name 'basestring' is not defined

danielehrhardt, 2021-12-28:
```
Traceback (most recent call last):
  File "/home/cnc/.local/bin/qtpyvcp", line 8, in <module>
    sys.exit(main())
  File "/home/cnc/.local/lib/python3.9/site-packages/qtpyvcp/app/__init__.py", line 73, in main
    opts = parse_opts(__doc__)
  File "/home/cnc/.local/lib/python3.9/site-packages/qtpyvcp/utilities/opt_parser.py", line 104, in parse_opts
    opts = DotDict({arg.strip('-<>').replace('-', '_') : convType(value) for arg, value in raw_args.items()})
  File "/home/cnc/.local/lib/python3.9/site-packages/qtpyvcp/utilities/opt_parser.py", line 104, in <dictcomp>
    opts = DotDict({arg.strip('-<>').replace('-', '_') : convType(value) for arg, value in raw_args.items()})
  File "/home/cnc/.local/lib/python3.9/site-packages/qtpyvcp/utilities/opt_parser.py", line 65, in convType
    if isinstance(val, basestring):
NameError: name 'basestring' is not defined

```

danielehrhardt, 2021-12-28:
```
cd ~/dev
git clone git://github.com/kcjengr/probe_basic.git
cd probe_basic
git checkout python3
qcompile .
python3 -m pip install --no-deps --editable .
cp -r ~/dev/probe_basic/probe_basic/fonts/ ~/.local/share/
cp -r ~/dev/probe_basic/config/probe_basic/ ~/linuxcnc/configs/
cd ~/dev
git clone git://github.com/kcjengr/qtpyvcp.conversational-gcode.git
cd qtpyvcp.conversational-gcode
git checkout python3
python3 -m pip install --no-deps --editable .
```

TurBoss, 2021-12-29:
hello,

could you post the output of  `qtpyvcp --info`

I'm not sure about the --no-deps option

if that is to skip installing qtpyvcp when installing probe_basic maybe because its already installed, yo can just delete Line 24
from setup.py

https://github.com/kcjengr/probe_basic/blob/python3/setup.py#L24

thanks

danielehrhardt, 2022-01-20:
> hello,
>
> could you post the output of `qtpyvcp --info`
>
> I'm not sure about the --no-deps option
>
> if that is to skip installing qtpyvcp when installing probe_basic maybe because its already installed, [...]

[image]

TurBoss, 2022-01-20:
Hello, 
for some reason the qtpyvcp version installed is the python2
but in in the python3 system

so maybe thats because `pip install qtpyvcp` that still points to python2

so maybe if you could try to uninstall qtpyvcp from pip and install from git

```
python3 -m pip uninstall  qtpyvcp # run more than once to be sure its uninstalled
cd ~/dev
git clone git://github.com/kcjengr/qtpyvcp.git
cd qtpyvcp
qcompile .
python3 -m pip install --editable .
```

joco-nz, 2023-05-05:
Closing as over a year old and reads as python version mixup.  Noting that Python 2 is now end of life.

closed 2023-05-05 by joco-nz