kcjengr/qtpyvcp #103: Traceback (most recent call last):   File "/home/dieter/.local/lib/python2.7/site-packages/qtpyvcp/widgets/input_widgets/tool_table.py", line 252, in saveToolTable     self.tool_model.saveToolTable()   File "/home/dieter/.local/lib/python2.7/site-packages/qtpyvcp/widgets/input_widgets/tool_table.py", line 200, in saveToolTable     self.tt.saveToolTable(self._tool_table, self._columns)   File "/home/dieter/.local/lib/python2.7/site-packages/qtpyvcp/plugins/tool_table.py", line 423, in saveToolTable     fh.write('\n'.join(lines)) TypeError: write() argument 1 must be unicode, not str

Herican, 2022-03-17:
1.  Ich kann in Probebasic keine Werkzeugtabelle(Tolls) speichern, wenn ich auf speichern gehe, kommt immer folgende Fehlermeldung
 

 Ich habe heute Probebasic neu, nach folgender vorgehensweise Installiert:
 https://kcjengr.github.io/probe_basic/quick_start.html

## Traceback

```python
Traceback (most recent call last):
  File "/home/dieter/.local/lib/python2.7/site-packages/qtpyvcp/widgets/input_widgets/tool_table.py", line 252, in saveToolTable
    self.tool_model.saveToolTable()
  File "/home/dieter/.local/lib/python2.7/site-packages/qtpyvcp/widgets/input_widgets/tool_table.py", line 200, in saveToolTable
    self.tt.saveToolTable(self._tool_table, self._columns)
  File "/home/dieter/.local/lib/python2.7/site-packages/qtpyvcp/plugins/tool_table.py", line 423, in saveToolTable
    fh.write('\n'.join(lines))
TypeError: write() argument 1 must be unicode, not str
```

## Options

```json
{
    "command_line_args": null, 
    "config_file": null, 
    "confirm_exit": null, 
    "develop": false, 
    "fullscreen": null, 
    "help": false, 
    "hide_cursor": false, 
    "hide_menu_bar": false, 
    "hide_status_bar": false, 
    "info": false, 
    "ini": "/home/dieter/linuxcnc/configs/Axis_Grundeinst_Original_ProbeB/CNC_Probe_Basic.ini", 
    "log_file": "/home/dieter/qtpyvcp.log", 
    "log_level": null, 
    "maximize": null, 
    "perfmon": false, 
    "position": null, 
    "qt_api": null, 
    "size": null, 
    "stylesheet": null, 
    "theme": null, 
    "version": false
}
```

## System Info
```
 * Description:	Debian GNU/Linux 10 (buster)
 * Kernel: 4.19.0-19-rt-amd64
 * Qt version: v5.11.3
 * Qt bindings: PyQt5 v5.11.3
 * LinuxCNC version: v2.8.2-11-g6a3d0a434
 * QtPyVCP version: v0.3.19+15.g3ad0cc9c
```

## Attachments

Please also find and attach the following files, along with any others that may be helpful:
* /home/dieter/qtpyvcp.log
* None

rmu75, 2023-03-17:
Leider ist das mit python 2.7 nicht 8-bit clean, vermutlich ist in der Tool-Tabelle ein Umlaut oder sonstiges nicht-ASCII Zeichen.

Mit der aktuellen Version basierend auf python 3 scheints jetzt auch mit Umlauten zu funktionieren.

rmu75, 2023-03-17:
Seems I misread the date, this bug is exactly one year old and probably no longer relevant. Sorry for exhuming this.

joco-nz, 2023-05-05:
Closing as old and part of 2.7 branch which is now defunct as Python 2 is end of life.

closed 2023-05-05 by joco-nz