kcjengr/qtpyvcp #109: Abstracting Controller - GrblHAL

eecue, 2022-05-27:
Following up on this ticket I filed:  https://github.com/kcjengr/probe_basic/issues/63

Ok reading the codebase I think the best bet is to totally abstract the controller. I see `linuxcnc` pretty much everywhere so I’m thinking it would make sense to totally abstract that as `controller` or something like that. This would even make it useable for something like Klipper down the road (no desire or intention to do that though lol) 

I’m going to take a first pass this weekend and share a PR.

andrewmarles, 2022-05-28:
Will help out if/where I can.

andrewmarles, 2022-06-05:
I have got the eRPC implementation to the point where I can start implementing more complex functions and structures.  The prototypes are not terribly flexible and so there will be a wrapper function required for pretty much everything in the GRBLHAL HAL, so it will be an iterative process to get everything in there.  For now I will start to make wrappers for the relevant machine, spindle and coolant actions:

[https://www.qtpyvcp.com/actions/index.html](https://www.qtpyvcp.com/actions/index.html)

One major piece of work will be to implement something in qtpyvcp that will stream a g-code program over the transport to GRBLHAL when the application expects LinuxCNC operating in AUTO mode.  I will add wrapper functions to enqueue g-code and check the IOStream buffer status, but a simple g-code sender will need to be implemented in python.

joco-nz, 2023-05-05:
Not sure why this is here.  I reads more like a comment meant for a fork. Either way this is not within the scope of the projects focus.

closed 2023-05-05 by joco-nz