kcjengr/qtpyvcp #112: TypeError: setMaximum(self, a0: int): argument 1 has unexpected type 'float'

smoe, 2022-12-27:
## Steps to reproduce the problem

Current master branch,
Debian unstable running python3.10.9,
all missing dependencies added from sources
manually copied configs from source tree
start linuxcnc
chosing sim.qtpyvcp 3axis metric
selecting action
see window with backtrace
followed invitation to submit issue

## It worked properly before this

First exploratory steps in QtPyVCP

## Traceback

```python
Traceback (most recent call last):
  File "/usr/bin/qtpyvcp", line 33, in <module>
    sys.exit(load_entry_point('qtpyvcp===python3-master-319.gfe86e6e6.dirty', 'console_scripts', 'qtpyvcp')())
  File "/usr/lib/python3/dist-packages/qtpyvcp/app/__init__.py", line 74, in main
    run(opts, None)
  File "/usr/lib/python3/dist-packages/qtpyvcp/app/__init__.py", line 102, in run
    load_vcp(opts)
  File "/usr/lib/python3/dist-packages/qtpyvcp/app/launcher.py", line 131, in load_vcp
    if _load_vcp_from_entry_point(vcp, opts):
  File "/usr/lib/python3/dist-packages/qtpyvcp/app/launcher.py", line 181, in _load_vcp_from_entry_point
    vcp.main(opts)
  File "/usr/lib/python3/dist-packages/examples/actions/__init__.py", line 23, in main
    qtpyvcp.app.run(opts, VCP_CONFIG_FILE)
  File "/usr/lib/python3/dist-packages/qtpyvcp/app/__init__.py", line 129, in run
    launch_application(opts, config)
  File "/usr/lib/python3/dist-packages/qtpyvcp/app/launcher.py", line 82, in launch_application
    app.initialiseWidgets()
  File "/usr/lib/python3/dist-packages/qtpyvcp/app/application.py", line 251, in initialiseWidgets
    w.initialize()
  File "/usr/lib/python3/dist-packages/qtpyvcp/widgets/input_widgets/setting_slider.py", line 145, in initialize
    self.setMaximum(self._setting.max_value)
TypeError: setMaximum(self, a0: int): argument 1 has unexpected type 'float'
```

## Options

```json
{
    "chooser": false,
    "command_line_args": null,
    "config_file": "/home/moeller/linuxcnc/configs/sim.qtpyvcp/custom_config.yml",
    "confirm_exit": false,
    "develop": false,
    "fullscreen": null,
    "help": false,
    "hide_cursor": false,
    "hide_menu_bar": false,
    "hide_status_bar": false,
    "info": false,
    "ini": "/home/moeller/linuxcnc/configs/sim.qtpyvcp/xyz-metric.ini",
    "log_file": "/home/moeller/linuxcnc/configs/sim.qtpyvcp/sim.log",
    "log_level": "WARN",
    "maximize": null,
    "perfmon": false,
    "position": null,
    "qt_api": null,
    "size": null,
    "stylesheet": null,
    "theme": null,
    "vcp": "actions",
    "version": false
}
```

## System Info
```
 * b'Description:\tDebian GNU/Linux bookworm/sid'
 * Kernel: b'6.0.0-6-rt-amd64'
 * Qt version: v5.15.7
 * Qt bindings: PyQt5 v5.15.7
 * LinuxCNC version: v2.9.0~pre0+git20221105.ffb6bda926
 * QtPyVCP version: python3_master+319.gfe86e6e6.dirty
```

## Attachments

Please also find and attach the following files, along with any others that may be helpful:
* /home/moeller/linuxcnc/configs/sim.qtpyvcp/sim.log
* /home/moeller/linuxcnc/configs/sim.qtpyvcp/custom_config.yml

TurBoss, 2022-12-27:
Hello, thanks for the report

I created a branch called "issue_112" where i try to fix this

looks like int is required instead of float

Thanks

TurBoss, 2022-12-29:
hello, I need to implement a slider that can handle floats

can take a bit :-)

thanks

TurBoss, 2022-12-29:
hello, since the data received by the slider are floats but all have 0 as decimal part the cast to int is a good solution for

thanks

smoe, 2022-12-29:
Addressed in https://github.com/kcjengr/qtpyvcp/commit/dc93f3938d9ed71f8bf5545a4c4b4b093e7cbb95 .

closed 2022-12-29 by smoe