kcjengr/qtpyvcp #117: ModuleNotFoundError: No module named 'vtkmodules'

flounderscore, 2023-03-01:
I'm trying to run probe_pasic_lathe on a freshly installed Debian 10, LinuxCNC 2.10 system.
Debian 10 comes with Python 3.7 which appears to be supported by qtpyvcp. However, the latest deb for vtk is `python3-vtk7`, and pip could install vtk8. No straight forward way to install `python3-vtk9` as written in the qtpyvcp documentation. When running probe_pasic_lathe, I get the following error:

```
  File "/home/flo/Documents/files/repositories/qtpyvcp/qtpyvcp/widgets/display_widgets/vtk_backplot/tool_actor.py", line 9, in <module>
    from vtkmodules.vtkCommonColor import vtkNamedColors
ModuleNotFoundError: No module named 'vtkmodules'
```

Is this because of the "wrong" vtk version? If that is the case, the documentation should be updated because there is no vtk9 for Python 3.7 as far as I can tell.

flounderscore, 2023-03-01:
Renaming `https://files.pythonhosted.org/packages/bf/ce/5227675482f49f4add76b91fd15be87e8e44155cbbefbe151119f8784825/vtk-9.2.6-cp37-cp37m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl` to `*manylinux1*` let me install vtk9 on Debian 10.

Lcvette, 2023-03-01:
You need Debian 11 or later to run python 3 probe basic and linuxcnc 2.9+

Lcvette, 2023-03-01:
> I'm trying to run probe_pasic_lathe on a freshly installed Debian 10, LinuxCNC 2.10 system. Debian 10 comes with Python 3.7 which appears to be supported by qtpyvcp. However, the latest deb for vtk [...]

You need Debian 11 or later to run python3 probe basic and linuxcnc 2.9+

Lcvette, 2023-03-01:
Mismatch version issue.

closed 2023-03-01 by Lcvette