kcjengr/qtpyvcp #130: AttributeError: module 'monokrom' has no attribute 'VCP_CONFIG_FILE'

kang2k10, 2023-12-01:
Linuxcnc 2.9 from official iso
install qtpyvcp from script "/usr/lib/python3/dist-packages/qtvcp/designer/install_script"
from this repo, clone example.
after run sim.qtpyvcp/xyz.ini
this error:
  File "/usr/lib/python3/dist-packages/qtpyvcp/vcp_chooser/vcp_chooser.py", line 79, in get_vcp_data
    vcp_config_file = vcp.VCP_CONFIG_FILE
                      ^^^^^^^^^^^^^^^^^^^
AttributeError: module 'monokrom' has no attribute 'VCP_CONFIG_FILE'

KurtJacobson, 2023-12-02:
I have this same issue on a fresh install of Monokrom from a deb package.

TurBoss, 2023-12-03:
same issue here looks it has to be with vcpchooser

kang2k10, 2023-12-21:
Sorry, i'm mixed QtVcp & QtPyVcp

closed 2023-12-21 by kang2k10

antarcticstar1, 2024-01-16:
How to solve this problem？

joco-nz, 2024-01-16:
Due to the changes in how Debian 12 and other recent distros treat non Debian packaged python use:

https://www.qtpyvcp.com/install/apt_install.html

Unless you are trying to do a developer install. In which case look at the relevant pages on the same site.