kcjengr/qtpyvcp #144: TypeError: a bytes-like object is required, not 'str'

copysent, 2024-07-08:
The error message I see:

`[qtpyvcp.app.launcher][DEBUG]  Initializing widgets (launcher.py:81)
Exception in thread Thread-1 (hal_poll_thread):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/qtpyvcp/utilities/obj_status.py", line 336, in hal_poll_thread
    raw = rawtuple[0].split('\n')
          ^^^^^^^^^^^^^^^^^^^^^^^
TypeError: a bytes-like object is required, not 'str'`

The location of the code in the tree:

https://github.com/kcjengr/qtpyvcp/blob/c20d4286cab764eb088c9e5e7fd34d94469cf3a5/src/qtpyvcp/utilities/obj_status.py#L336

Example code for testing:

```
from qtpyvcp.utilities.obj_status import HALStatus
hal_status = HALStatus()

def test_def(value):
    pass

pin = hal_status.getHalPin('some.existing.pin')

pin.connect(test_def)
```

smoe, 2025-03-17:
I have no idea why this issue was not yet addressed. Maybe forgotten? Is it still of convern in the latest version? I just checked - yes, it is:

https://github.com/kcjengr/qtpyvcp/blob/main/src/qtpyvcp/utilities/obj_status.py#L336

I asked chatGPT about this error, and it answered very nicely. With that information, maybe you can attempt to patch this yourself, test it (which is where the work goes) and prepare a pull request with a confirmation that this fixes your issue?

```
This indicates that rawtuple[0] is a bytes object, not a string (str).

In Python 3, bytes and str are distinct types.
The .split('\n') method expects a string delimiter when used on a str object, but if you're working with bytes, you need to provide a bytes delimiter as well.
How to fix it:

Option 1: Decode the bytes to a string first (common approach):

raw = rawtuple[0].decode('utf-8').split('\n')
Use .decode() to convert bytes to str.
Choose the correct encoding (utf-8 is typically safe).

Option 2: Split using a bytes delimiter (less common):

raw = rawtuple[0].split(b'\n')
Only makes sense if you intend to keep working with bytes rather than decoded strings.

Summary
Cause: You have a bytes object and are using a str delimiter.
Solution: Either decode to a string, or use a bytes delimiter (b'\n')
```

TurBoss, 2025-03-18:
Hello,

I took a look and tried the fixes but more issues followed, it can take bit of time :-)

Thanks for the reminder

TurBoss, 2025-03-18:
That fixed the error messages but I cant get the callback to print.

I ran this while linuxcnc is running

```python
from qtpyvcp.utilities.obj_status import HALStatus
hal_status = HALStatus()

def test_def(value):
    print(value)

pin = hal_status.getHALPin("joint.0.pos-cmd")

pin.connect(test_def)
```

smoe, 2025-03-18:
And maybe also in https://github.com/kcjengr/qtpyvcp/blob/1a6d27f4418211800241c16144afceae297462f6/src/qtpyvcp/utilities/obj_status.py#L338
a `split(b' ')`?