kcjengr/qtpyvcp #153: Request to change Debian package install dir of examples

smoe, 2025-03-21:
The examples directory is too generic, the tool `lintian` (checking the Debian packaging) complains:
```
E: python3-qtpyvcp: python-module-has-overly-generic-name (examples) [usr/lib/python3/dist-packages/examples/__init__.py]
```
and it has a point. I have no idea if that path is referenced from anywhere, so I cannot tell if that is any difficult to change. I suggest to rename the "examples" folder to qtpyvcp-examples.

TurBoss, 2025-03-21:
hello,

I pushed the changes to a the `package_names` branch

will merge to main asap

smoe, 2025-03-28:
Should the examples possibly be distributed in a separate package?

TurBoss, 2025-03-29:
Yes i think that may be posible